varigit/variscite-bsp-platform #5: Invalid hash in manifest

csonsino, 2021-01-21:
The `fsl-zeus` branch contains an invalid hash in the `imx-5.4.47-2.2.0-var01.xml` manifest for the `meta-nxp-demo-experience` repo:

https://github.com/varigit/variscite-bsp-platform/blob/c1e4d26bb268f9da677e6037469a04064db42818/imx-5.4.47-2.2.0-var01.xml#L37

Please update this file with a valid hash.  I'm not sure if the latest is "community compatible", but I'm currently testing with `09a7201ee64d09202b3546fef8ea34bb671cb1cd`.

eranmati, 2021-01-22:
Hi @csonsino.
It is actually valid, and is the latest commit of the zeus-5.4.24-2.1.0 branch:
https://source.codeaurora.org/external/imxsupport/meta-nxp-demo-experience/commit/?h=zeus-5.4.24-2.1.0&id=c7263d9f3cc7bbf44e7164ffeda494cf283d3dec

I've also just verified that there are no issues running "repo init" and "repo sync" with this manifest.

Regards,
Eran

closed 2021-01-22 by eranmati

csonsino, 2021-01-22:
Oh, my bad!  Thanks for the pointer - I had a bad mirror and was pointing at the [nxp-demo-experience source](https://source.codeaurora.org/external/imxsupport/nxp-demo-experience/) instead of the meta-nxp-demo-experience layer.