varigit/variscite-bsp-platform #11: Error Installing BSP Platform (Syntax Error)

ChaosCzar, 2022-03-02:
Upon attempting to install on a system running Ubuntu 16.04, the following syntax error occurs.

carter@carter-VirtualBox:~/var-fsl-yocto$ repo init -u https://github.com/varigit/variscite-bsp-platform.git -b pyro
Traceback (most recent call last):
  File "/home/carter/var-fsl-yocto/.repo/repo/main.py", line 42, in <module>
    from git_config import RepoConfig
  File "/home/carter/var-fsl-yocto/.repo/repo/git_config.py", line 774
    self._Set(f'superproject.{key}', value)
                                  ^
SyntaxError: invalid syntax

david-sterling, 2022-03-02:
Hi! Are you sure you had installed python-git packet? Pyro is quite old, I cannot remember if you need also ancient https://pypi.org/project/gitconfig/ packet.

nsdrude, 2022-03-02:
@ChaosCzar You need to install Python 3.6. You can refer to Variscite's Ubuntu 16.04 Dockerfile as an example: https://github.com/varigit/var-host-docker-containers/blob/4b27233bfd54630d2c81ea6b237ca2959f7bd31d/Dockerfile_16.04#L76-L83

closed 2022-03-02 by nsdrude