varigit/variscite-bsp-platform #13: Kirkstone branch chooses wrong kernel for `imx6ul-var-dart`

darkdragon-001, 2022-11-17:
When building the `kirkstone` branch for `imx6ul-var-dart`, it chooses `linux-variscite` kernel version `4.14` from `meta-freescale-3rdparty` repository instead of `5.15` from `meta-variscite-bsp` repository.

When I try to force the version via `REQUIRED_VERSION_linux-variscite = "5.15%"` in my distro, it says this version is not available.

What can go wrong here?

darkdragon-001, 2022-11-17:
When will it be available? We need newer toolchain, kernel and packages as we use the Yocto SDK to build our own software for this device. As we share code with other devices, this is the major blocker in using newer features like C++20 in our shared code...

nsdrude, 2022-11-17:
@darkdragon-001 please contact sales@variscite.com regarding the software roadmap.

moto-timo, 2023-03-03:
I have submitted https://github.com/varigit/meta-variscite-bsp/pull/17 for https://github.com/varigit/meta-variscite-bsp/issues/10 which should resolve this for `kirkstone` and `imx6` platforms. Since I only have access to `DART-6UL` hardware, I cannot verify that it is functional on `VAR-SOM-6UL` (but it should be).

closed 2023-12-18 by nsdrude