varigit/variscite-bsp-platform #14: Replacing the Linux logo kernel_conf_variable not found

O2SHEN, 2022-12-05:
Hello,

I have bellow issue when replacing the Linux logo follow instruction https://variwiki.com/index.php?title=Yocto_Linux_logo:
var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+45f7993209-r0/temp/run.do_configure.5106: kernel_conf_variable: not found

It looks like the kernel_conf_variable function invar-fsl-yocto/sources/meta-openembedded/meta-oe/recipes-kernel/linux/linux.inc is missing.

dev machine:ubuntu 18.04
source code branch: kirkstone (latest)

varigigi, 2022-12-07:
Hi @O2SHEN,
according the manifest, you should get the following code
https://github.com/openembedded/meta-openembedded/blob/kirkstone/meta-oe/recipes-kernel/linux/linux.inc#L27-L35
Please ensure you are picking the right version of meta-openembedded layer.
Thanks
Best Regards
Pier

O2SHEN, 2022-12-08:
Hi @varigigi .

Thanks for your quick response. 
As you point out, I checked the lastest manifest https://github.com/varigit/variscite-bsp-platform/blob/kirkstone/kirkstone-5.15.xml, and it shows that the version of meta-openembedded layer it used is 744a4b6eda88b9a9ca1cf0df6e18be384d9054e3 of branch kirkstone. And these lines you specified is there, but the weird thing is it keep reported "kernel_conf_variable: not found". At last I copy those lines into my do_install function and it works for me now!

Thank you so much
Sam

closed 2025-03-26 by nsdrude-var