varigit/variscite-bsp-platform #15: error when building yocto from source code for DART-MX8M-PLUS - Yocto Kirkstone 4.0 based on FSL Community BSP 4.0 with L5.15.32_2.0.0 Linux release

bahaeg, 2023-03-15:
I'm trying to build the fsl-image-gui for the dart-mx8m-plus board using the latest Kirkstone release using the [build Yocto from source code](https://variwiki.com/index.php?title=Yocto_Build_Release&release=mx8mp-yocto-kirkstone-5.15-2.0.x-v1.2).

I'm using ubuntu 22.04 as a dev machine.
The log files of the error shows: /usr/bin/env: ‘node’: No such file or directory. But when I execute /usr/bin/env node --version, node is found.
Please find below the full error message.

``` sh
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mp-var-dart"
DISTRO               = "fslc-xwayland"
DISTRO_VERSION       = "4.0"
TUNE_FEATURES        = "aarch64 armv8a crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:24a3f7b3648185e33133f5d96b184a6cb6524f3d"
meta-oe              
meta-multimedia      
meta-python          
meta-filesystems     
meta-gnome           
meta-networking      = "HEAD:744a4b6eda88b9a9ca1cf0df6e18be384d9054e3"
meta-freescale       = "HEAD:3747641f1e71d8e4edd5b587b49d09dc2d243942"
meta-freescale-3rdparty = "HEAD:6d8213fc5ec192c33f963b8095d1f01af1574eea"
meta-freescale-distro = "HEAD:d5bbb487b2816dfc74984a78b67f7361ce404253"
meta-freescale-ml    = "HEAD:08dcb7c67072907348cd235d9e110d4ba91e8235"
meta-clang           = "HEAD:c728c3f9168c8a4ed05163a51dd48ca1ad8ac21d"
meta-chromium        = "HEAD:e232c2e21b96dc092d9af8bea4b3a528e7a46dd6"
meta-qt5             = "HEAD:44d44933200287f7d17cf6981af4b4a0961c308d"
meta-python2         = "HEAD:f02882e2aa9279ca7becca8d0cedbffe88b5a253"
meta-swupdate        = "HEAD:fabbcc623d41794fbfa97c38b9da941ed2cdcdcc"
meta-virtualization  = "HEAD:5be54854139acb4796000478394afd66de0792c1"
meta-variscite-bsp   = "HEAD:838e550f2bcc19daae68c2e856bc32d55333b871"
meta-variscite-sdk   = "HEAD:85ac99485dbe17f8e63782451afe7814901dc8e5"
meta-variscite-hab   = "HEAD:4acacfdbc4b89ace606b5d640e768b7561435324"

Initialising tasks: 100% |##################################################| Time: 0:00:03
Sstate summary: Wanted 233 Local 172 Mirrors 0 Missed 61 Current 4028 (73% match, 98% complete)
NOTE: Executing Tasks
ERROR: linux-variscite-5.15.60+gitAUTOINC+823c33d95f-r0 do_kernel_configme: ExecutionError('/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/temp/run.do_kernel_configme.3547021', 1, None, None)
ERROR: Logfile of failure stored in: /home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/temp/log.do_kernel_configme.3547021
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/work/var-fsl-yocto/sources/poky/meta/recipes-devtools/binutils/binutils-cross_2.38.bb:do_populate_sysroot', '/home/work/var-fsl-yocto/sources/poky/meta/recipes-devtools/gcc/gcc-cross_11.3.bb:do_populate_sysroot', '/home/work/var-fsl-yocto/sources/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/work/var-fsl-yocto/sources/poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/home/work/var-fsl-yocto/sources/poky/meta/recipes-devtools/bison/bison_3.8.2.bb:do_populate_sysroot', 'virtual:native:/home/work/var-fsl-yocto/sources/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/work/var-fsl-yocto/sources/poky/meta/recipes-extended/bc/bc_1.07.1.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['binutils-cross-aarch64', 'gcc-cross-aarch64', 'quilt-native', 'kern-tools-native', 'bison-native', 'patch-native', 'bc-native', 'gettext-minimal-native', 'flex-native', 'texinfo-dummy-native', 'xz-native', 'libtool-native', 'gnu-config-native', 'zlib-native', 'zstd-native', 'linux-libc-headers', 'gmp-native', 'libmpc-native', 'mpfr-native', 'attr-native', 'readline-native', 'm4-native', 'ncurses-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_kernel_configme
| NOTE: do_kernel_metadata: for summary/debug, set KCONF_AUDIT_LEVEL > 0
| /usr/bin/env: ‘node’: No such file or directory
| /usr/bin/env: ‘node’: No such file or directory
| /usr/bin/env: ‘node’: No such file or directory
| /usr/bin/env: ‘node’: No such file or directory
| /usr/bin/env: ‘node’: No such file or directory
| NOTE: kernel meta data summary for imx8mp-var-dart (standard):
| NOTE: ======================================================================
| NOTE: BSP entry point / definition:
| NOTE: KBUILD_DEFCONFIG: imx8_var_defconfig
| NOTE: Fragments from SRC_URI:
| NOTE: KERNEL_FEATURES:
| NOTE: Final scc/cfg list: /home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/defconfig
| WARNING: /home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/temp/run.do_kernel_configme.3547021:175 exit 1 from 'CFLAGS=" -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0=/usr/src/debug/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0                      -fdebug-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0=/usr/src/debug/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0                      -fdebug-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native=   --sysroot=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot" HOSTCC="gcc  -isystem/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native/usr/lib                         -L/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/work/var-fsl-yocto/build_xwayland/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2" HOSTCPP="gcc  -E" CC="aarch64-fslc-linux-gcc   -fuse-ld=bfd -fmacro-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0=/usr/src/debug/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0                      -fdebug-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0=/usr/src/debug/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0                      -fdebug-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/linux-variscite/5.15.60+gitAUTOINC+823c33d95f-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/work/var-fsl-yocto/build_xwayland/tmp/work/imx8mp_var_dart-f [... cut ...]
ERROR: Task (/home/work/var-fsl-yocto/sources/meta-variscite-bsp/recipes-kernel/linux/linux-variscite_5.15.bb:do_kernel_configme) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3079 tasks of which 3078 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds

Summary: 1 task failed:
  /home/work/var-fsl-yocto/sources/meta-variscite-bsp/recipes-kernel/linux/linux-variscite_5.15.bb:do_kernel_configme
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

Xenoamor, 2023-03-15:
I can't say why yours has failed but I'm using the following to build from within docker

```
git clone https://github.com/varigit/var-host-docker-containers.git
cd var-host-docker-containers
./run.sh -u 22.04

mkdir docker-yocto-build
cd docker-yocto-build
repo init -u https://github.com/varigit/variscite-bsp-platform.git -b refs/tags/mx8mp-yocto-kirkstone-5.15-2.0.x-v1.2 -m default.xml
repo sync -j4

MACHINE=imx8mp-var-dart DISTRO=fslc-xwayland source ./var-setup-release.sh build_xwayland
bitbake fsl-image-gui
```

nsdrude, 2024-04-24:
@bahaeg please open a ticket in our customer support portal if you need additional support on this.

Thanks,
Nate

closed 2024-04-24 by nsdrude