varigit/variscite-bsp-platform #16: Error building kirkstone-5.15.xml

esguti, 2023-03-16:
I just followed the steps explained in [Variscite wiki for building kirkstone](https://variwiki.com/index.php?title=Yocto_Build_Release&release=mx8mp-yocto-kirkstone-5.15-2.0.x-v1.2)

When I execute `bitbake fsl-image-gui` , a mismatch version for **isp-imx-4.2.2.18.0.bin** appears and I have to stop the compilation because it does not finish well.

```
NOTE: Your conf/bblayers.conf has been automatically updated.
WARNING: You have included the meta-virtualization layer, but 'virtualization' has not been enabled in your DISTRO_FEATURES. Some bbappend files may not take effect. See the meta-virtualization README for details on enabling virtualization support.
Loading cache: 100% |                                                                                                                                                                              | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |#############################################################################################################################################################################| Time: 0:00:42
Parsing of 3426 .bb files complete (0 cached, 3426 parsed). 5177 targets, 569 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mp-var-dart"
DISTRO               = "fslc-xwayland"
DISTRO_VERSION       = "4.0"
TUNE_FEATURES        = "aarch64 armv8a crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:24a3f7b3648185e33133f5d96b184a6cb6524f3d"
meta-oe              
meta-multimedia      
meta-python          
meta-filesystems     
meta-gnome           
meta-networking      = "HEAD:744a4b6eda88b9a9ca1cf0df6e18be384d9054e3"
meta-freescale       = "HEAD:3747641f1e71d8e4edd5b587b49d09dc2d243942"
meta-freescale-3rdparty = "HEAD:6d8213fc5ec192c33f963b8095d1f01af1574eea"
meta-freescale-distro = "HEAD:d5bbb487b2816dfc74984a78b67f7361ce404253"
meta-freescale-ml    = "HEAD:08dcb7c67072907348cd235d9e110d4ba91e8235"
meta-qt5             = "HEAD:44d44933200287f7d17cf6981af4b4a0961c308d"
meta-python2         = "HEAD:f02882e2aa9279ca7becca8d0cedbffe88b5a253"
meta-swupdate        = "HEAD:fabbcc623d41794fbfa97c38b9da941ed2cdcdcc"
meta-virtualization  = "HEAD:5be54854139acb4796000478394afd66de0792c1"
meta-variscite-bsp   = "HEAD:fa2757a53d2f9c3283c2de4d3142a51b1fa0e3ae"
meta-variscite-sdk   = "HEAD:5664513a0cf0f43afb60f92e63e60a86f3e53bd0"
meta-variscite-hab   = "HEAD:4acacfdbc4b89ace606b5d640e768b7561435324"

NOTE: Fetching uninative binary shim http://downloads.yoctoproject.org/releases/uninative/3.7/x86_64-nativesdk-libc-3.7.tar.xz;sha256sum=b110bf2e10fe420f5ca2f3ec55f048ee5f0a54c7e34856a3594e51eb2aea0570 (will check PREMIRRORS first)
Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:08
Sstate summary: Wanted 4019 Local 0 Mirrors 0 Missed 4019 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
WARNING: cracklib-2.9.8-r0 do_fetch: Failed to fetch URL git://github.com/cracklib/cracklib;protocol=https;branch=master, attempting MIRRORS if available
WARNING: vulkan-headers-1.2.182.0-r0 do_fetch: Failed to fetch URL git://github.com/KhronosGroup/Vulkan-Headers.git;branch=master;protocol=https, attempting MIRRORS if available
WARNING: vulkan-loader-1.2.182.0-r0 do_fetch: Failed to fetch URL git://github.com/KhronosGroup/Vulkan-Loader.git;protocol=https;branch=master, attempting MIRRORS if available
WARNING: quota-4.06-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/project/linuxquota/quota-tools/4.06/quota-4.06.tar.gz, attempting MIRRORS if available
WARNING: assimp-5.0.1-r0 do_fetch: Failed to fetch URL git://github.com/assimp/assimp.git;protocol=https;branch=assimp_5.0_release, attempting MIRRORS if available
WARNING: isp-imx-4.2.2.18.0-r0 do_fetch: Checksum mismatch for local file /home/yocto/yocto-variscite/downloads/isp-imx-4.2.2.18.0.bin
Cleaning and trying again.
WARNING: isp-imx-4.2.2.18.0-r0 do_fetch: Renaming /home/yocto/yocto-variscite/downloads/isp-imx-4.2.2.18.0.bin to /home/yocto/yocto-variscite/downloads/isp-imx-4.2.2.18.0.bin_bad-checksum_e838c306e41fc8e4636e787ef510b1a32479999059b6a61a08d75b0aca6d24a3
WARNING: isp-imx-4.2.2.18.0-r0 do_fetch: Checksum failure encountered with download of https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//isp-imx-4.2.2.18.0.bin;fsl-eula=true - will attempt other sources if available
ERROR: isp-imx-4.2.2.18.0-r0 do_fetch: Fetcher failure for URL: 'https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//isp-imx-4.2.2.18.0.bin;fsl-eula=true'. Checksum mismatch!
File: '/home/yocto/yocto-variscite/downloads/isp-imx-4.2.2.18.0.bin' has md5 checksum 'bf740f1a712c59ea42f5b1a16ca147e4' when 'a20171db4bf2be423a587f3b610f0a69' was expected
File: '/home/yocto/yocto-variscite/downloads/isp-imx-4.2.2.18.0.bin' has sha256 checksum 'e838c306e41fc8e4636e787ef510b1a32479999059b6a61a08d75b0aca6d24a3' when '468ae51223d1873a1a756a1e64a53c0c61ebd640b3810f3a9e912b6a0de6c3c8' was expected
If this change is expected (e.g. you have upgraded to a new version without updating the checksums) then you can use these lines within the recipe:
SRC_URI[sha256sum] = "e838c306e41fc8e4636e787ef510b1a32479999059b6a61a08d75b0aca6d24a3"
Otherwise you should retry the download and/or check with upstream to determine if the file has become corrupted or otherwise unexpectedly modified.
ERROR: isp-imx-4.2.2.18.0-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'https://www.nxp.com/lgfiles/NMG/MAD/YOCTO//isp-imx-4.2.2.18.0.bin;fsl-eula=true')
ERROR: Logfile of failure stored in: /home/yocto/yocto-variscite/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/isp-imx/4.2.2.18.0-r0/temp/log.do_fetch.1881241
ERROR: Task (/home/yocto/yocto-variscite/sources/meta-freescale/recipes-bsp/isp-imx/isp-imx_4.2.2.18.0.bb:do_fetch) failed with exit code '1'

Second Keyboard Interrupt, stopping...

Summary: 1 task failed:
  /home/yocto/yocto-variscite/sources/meta-freescale/recipes-bsp/isp-imx/isp-imx_4.2.2.18.0.bb:do_fetch
Summary: There were 9 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
```

nsdrude, 2023-03-16:
@esguti thanks for reporting this. It looks like NXP changed the file without changing it's revision.

We'll push a fix soon. In the meantime, you can manually change meta-freescale/recipes-bsp/isp-imx/isp-imx_4.2.2.18.0.bb to align with the following fix from NXP: https://github.com/nxp-imx/meta-imx/commit/9fa550c958b950526e1bb1c8e8ed8b6ac4f29119

esguti, 2023-03-18:
Thank you for your quick response, problem is solved

closed 2023-03-18 by esguti