varigit/variscite-bsp-platform #17: codeaurora.org downloads broken

joelsand, 2023-03-31:
Hi there!

According to https://bye.codeaurora.org/, it's shutdown as of today so downloads that reference source.codeaurora.org are failing (see example below).

It looks like IMX has updated their recipes (e.g. https://github.com/Freescale/meta-freescale/commit/83b779a7ef69006564fb8d27e65c31917bc3b6f0). Any ETA on when the Variscite BSP file(s) will be updated?

Thanks,
Joel

```
ERROR: gstreamer1.0-plugins-good-1.20.0.imx-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'gitsm://source.codeaurora.org/external/mx8-fslc-linux/gstreamer1.0-plugins-good/1.20.0.imximx/gst-plugins-good.git;protocol=https;branch=MM_04.07.00_2205_L5.15.y')
```

nsdrude, 2023-03-31:
@joelsand thanks for reporting this. This will be fixed moving forward. 

For existing releases, it can be fixed by adding MIRRORS to local.conf. Please see: https://variwiki.com/index.php?title=Yocto_Common_Errors#codeaurora.org_migration

joelsand, 2023-03-31:
Thank you @nsdrude - that appears to be working!

closed 2023-12-18 by nsdrude