varigit/variscite-bsp-platform #18: build fails on nodejs for yocto version kirkstone-5.15.71_2.2.0-v1.0

ThomasKikkert, 2023-04-11:
# build steps

the development environment is ubuntu 22.04 LTS

followed the steps on [yocto build](https://variwiki.com/index.php?title=Yocto_Build_Release&release=mx8mp-yocto-kirkstone-5.15.71_2.2.0-v1.0)

```bash
mkdir var-fsl-yocto
cd var-fsl-yocto
sudo apt update
sudo apt upgrade -y
sudo apt install -y gawk wget git diffstat unzip texinfo gcc-multilib \
    build-essential chrpath socat cpio python python3 python3-pip python3-pexpect \
    xz-utils debianutils iputils-ping libsdl1.2-dev xterm libyaml-dev libssl-dev \
    autoconf libtool libglib2.0-dev libarchive-dev \
    sed cvs subversion coreutils texi2html docbook-utils python-pysqlite2 \
    help2man make gcc g++ desktop-file-utils libgl1-mesa-dev libglu1-mesa-dev \
    mercurial automake groff curl lzop asciidoc u-boot-tools dos2unix mtd-utils pv \
    libncurses5 libncurses5-dev libncursesw5-dev libelf-dev zlib1g-dev bc rename \
    zstd libgnutls28-dev python3-git zstd liblz4-tool python-is-python3 
    
repo init -u https://github.com/varigit/variscite-bsp-platform.git -b refs/tags/mx8mp-yocto-kirkstone-5.15.71_2.2.0-v1.0 -m kirkstone-5.15.71-2.2.0.xml
repo sync -j$(nproc)
MACHINE=imx8mp-var-dart DISTRO=fslc-xwayland . var-setup-release.sh build_xwayland
bitbake fsl-image-gui
```

## change in config -> local.conf
[buildlogNodejs.txt](https://github.com/varigit/variscite-bsp-platform/files/11200229/buildlogNodejs.txt)

```txt
MIRRORS += " \
    git://source.codeaurora.org/external/imx/ git://github.com/nxp-imx/;protocol=https \
    https://source.codeaurora.org/external/imx/ https://github.com/nxp-imx/ \
    http://source.codeaurora.org/external/imx/ http://github.com/nxp-imx/ \
    gitsm://source.codeaurora.org/external/imx/ gitsm://github.com/nxp-imx/;protocol=https \
    git://source.codeaurora.org/external/imx/eiq-tvm-imx git://github.com/nxp-imx/eiq-tvm-imx \
    git://source.codeaurora.org/external/imx/nn-imx.git git://github.com/nxp-imx/nn-imx.git \
"

BB_NUMBER_THREADS = "2"
```

ThomasKikkert, 2023-04-11:
I'm building the yocto image using 4 core with a total of 4 threads, 13GB of ram in VMware player 17 and 340GB of storage, running a maximum of 2 tasks at ones as recommend required amount of RAM 8GB. I ran into memory issue during compellation of some other layer, before limiting the thread count. As descript in the [common errors](https://variwiki.com/index.php?title=Yocto_Common_Errors#Hardware_resources) that fixed it, but the build error for nodejs stayed.

I tried to clean built of that package:
```bash
bitbake fsl-image-gui -c clean nodejs
bitbake fsl-image-gui -c package nodejs
```
This had no effect.

nsdrude, 2023-04-11:
@ThomasKikkert your virtual machine doesn't have enough memory to build nodejs, it is very memory intensive. You may want to also limit [PARALLEL_MAKE](https://docs.yoctoproject.org/4.0.8/ref-manual/variables.html?highlight=parallel_make#term-PARALLEL_MAKE).

If this doesn't work, please try using a dedicated machine with enough memory. It is recommended to use a dedicated machine instead of virtual machines. Virtual machines are not efficient with the CPU virtualization and Disk IO.

In the future, please open a ticket using Variscite's customer support portal: https://www.variscite.com/support/

nsdrude, 2024-04-24:
@ThomasKikkert assuming this is resolved, I'm closing this issue. Please open a ticket in our customer portal for further support.

Thanks,
Nate