varigit/variscite-bsp-platform #22: Issue compiling Mickledore (kernel 6.1.36)

hlarouss, 2024-03-01:
Hello,

I cloned the repo following the userguide located there: https://variwiki.com/index.php?title=Yocto_Build_Release&release=mx8mp-yocto-mickledore-6.1.36_2.1.0-v1.1

Unfortunately, when I run bitbake (with any of the three proposed versions), I get the following error:

WARNING: Duplicate inclusion for mypath/var-fsl-yocto/sources/meta-clang/conf/nonclangable.conf in mypath/var-fsl-yocto/sources/meta-clang/conf/layer.conf
WARNING: Duplicate inclusion for mypath/var-fsl-yocto/sources/meta-clang/conf/nonscanable.conf in mypath/var-fsl-yocto/sources/meta-clang/conf/layer.conf
WARNING: Duplicate inclusion for mypath/var-fsl-yocto/sources/meta-virtualization/conf/distro/include/virt_security_flags.inc in mypath/var-fsl-yocto/sources/meta-virtualization/conf/layer.conf
ERROR: Found duplicated BBFILE_COLLECTIONS 'filesystems-layer', check bblayers.conf or layer.conf to fix it.

It seems like there are some duplicate recipes inclusion in the bblayers.conf file.

If, instead, I use the bblayers.conf file obtained from "Mickledore (kernel 6.1.22)", the compilation seems to go through.

Am I doing something wrong here?

Thanks in advance for your help,
Regards,
Houssem

nsdrude, 2024-03-01:
@hlarouss please write the full steps to reproduce, starting with the repo init command.

hlarouss, 2024-03-01:
I'm literally following the user guide.

In ~:

```bash
sudo apt-get update && sudo apt-get dist-upgrade

sudo apt-get install gawk wget git diffstat unzip texinfo gcc-multilib \
 build-essential chrpath socat cpio python3 python3-pip python3-pexpect \
 xz-utils debianutils iputils-ping libsdl1.2-dev xterm libyaml-dev libssl-dev

sudo apt-get install autoconf libtool libglib2.0-dev libarchive-dev \
 sed cvs subversion coreutils texi2html docbook-utils \
 help2man make gcc g++ desktop-file-utils libgl1-mesa-dev libglu1-mesa-dev \
 mercurial automake groff curl lzop asciidoc u-boot-tools dos2unix mtd-utils pv \
 libncurses5 libncurses5-dev libncursesw5-dev libelf-dev zlib1g-dev bc rename \
 zstd libgnutls28-dev

sudo apt-get install python3-git zstd liblz4-tool

sudo apt-get install python-is-python3

mkdir var-fsl-yocto
cd var-fsl-yocto

repo init -u https://github.com/varigit/variscite-bsp-platform.git -b mickledore -m imx-6.1.36-2.1.0.xml
repo sync -j$(nproc)

MACHINE=imx8mp-var-dart DISTRO=fsl-imx-xwayland . var-setup-release.sh build_xwayland

bitbake fsl-image-gui-chromium
```

And I get the error mentioned above. I don't get it with Mickledore 6.1.22 or Kirkstone.

I'm on Pop! OS 22.04 which is 99% Ubuntu-22.04 based.

nsdrude, 2024-03-01:
@hlarouss I could not reproduce this using a new Ubuntu 22.04 virtual machine:
```
$ multipass launch 22.04 --name myubuntu22 --cpus 4 --memory 12G --disk 200G
$ multipass shell myubuntu22
```
Please try using Ubuntu instead of Pop! OS.

dorta, 2024-04-24:
Hey @hlarouss,

Since we do not support the Pop! OS 22.04 on Yocto, you can try to use the multipass as Nate wrote to you, or you can try to use Docker as described here:

https://variwiki.com/index.php?title=Docker_Build_Environment

It is very simple, and it might help you achieve the building you need.
If you need any help, please let us know.

Thanks,
Diego

dorta, 2024-04-24:
Hey @hlarouss,

I'm closing this issue on GitHub, if you need further support on this, please open another ticket on the portal.

Thanks,
Diego

closed 2024-04-24 by dorta