varigit/variscite-bsp-platform #23: Broken repo manifest in android-11.0.0_2.2.0-var01

berdroid, 2024-04-23:
Commit 39e5175 on android-11.0.0_2.2.0-var01 seems to have broken the repo manifest var-imx-android-11.0.0_2.2.0.xml

repo sync result is error
fatal: remote var-github not defined in /home/berndk/var_imx-android-11.0.0_2.2.0/android_build/.repo/manifest.xml

likely because the "var-github" remot was removed.

dorta, 2024-04-23:
Hey @berdroid,

I will take a look and try to fix this as soon as possible.

Thanks for noticing!

berdroid, 2024-04-24:
The following patch seems to make the repo tool happy again, but I cannot vouch for correctness:

    diff --git a/var-imx-android-11.0.0_2.2.0.xml b/var-imx-android-11.0.0_2.2.0.xml
    index 9946d43..5f44ddc 100644
    --- a/var-imx-android-11.0.0_2.2.0.xml
    +++ b/var-imx-android-11.0.0_2.2.0.xml
    @@ -8,6 +8,7 @@
     <remote name="imx-proprietary" fetch="https://github.com/nxp/" review="" />
     <remote name="wireless-regdb" fetch="https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/" review="" />
     <remote name="github" fetch="https://github.com/" review="" />
    +<remote name="var-github" fetch="https://github.com/varigit/" review="" />

     
     <project path="vendor/variscite/kernel_imx" name="linux-imx" remote="var-github" revision="android-11.0.0_2.2.0-var01" />

dorta, 2024-04-24:
Hey @berdroid,

I already fixed the issue, and I create a new PR for review.
It will be merged as soon as possible to the varigit GitHub account.

Thanks for reporting, and thanks for the patch suggestion.

dorta, 2024-04-24:
Hey @berdroid,

It's done: https://github.com/varigit/variscite-bsp-platform/commits/android-11.0.0_2.2.0-var01

Thanks,
Diego

closed 2024-04-24 by dorta