varigit/variscite-bsp-platform #24: meta-variscite-bsp-imx depends on inexistant and unsourced meta-fsl-bsp-release

valentinlageard, 2024-07-03:
On the mickledore branch, meta-variscite-bsp-imx depends on meta-fsl-bsp-release which is not referenced and sourced in the manifest. In addition, https://github.com/nxp-imx/meta-fsl-bsp-release seems abandoned and has no support for kernel with version beyond 4...

I guess the relevant layer has been moved somewhere but then which it is ? And layer.conf should be updated accordingly.

nsdrude, 2024-10-25:
@valentinlageard actually, fsl-bsp-release is required: https://github.com/nxp-imx/meta-imx/blob/mickledore-6.1.36-2.1.0/meta-bsp/conf/layer.conf#L8

Which is included in Variscite's bblayers.conf: 
- https://github.com/varigit-dev/meta-variscite-sdk-imx/blob/scarthgap_6.6.23-2.0.0_var01/scripts/var-setup-release.sh#L41
- https://github.com/nxp-imx/meta-imx/blob/mickledore-6.1.36-2.1.0/tools/imx-setup-release.sh#L161

jadinm, 2024-12-16:
Hello :-)

Are you sure that the issue did not require any action ?
Because from what I understand since building fails on scarthgap (and likely in mickledore) because of the non-existent layer `meta-fsl-bsp-release`.
I tried following the setup procedure documented in [README.md](https://github.com/varigit/variscite-bsp-platform/tree/scarthgap) and end up with the following error:

```shell
# <Same commands as in README>
$ MACHINE=imx8mm-var-dart DISTRO=fslc-framebuffer source ./var-setup-release.sh build
# <License agreement>
$ bitbake core-image-minimal                                                         
NOTE: Your conf/bblayers.conf has been automatically updated.
ERROR: Layer 'variscite-bsp' depends on layer 'fsl-bsp-release', but this layer is not enabled in your configuration
```

And indeed, there is no layer `fsl-bsp-release`:
```
$ ls sources 
base          meta-freescale           meta-matter               meta-security              meta-variscite-bsp-imx     meta-virtualization
meta-arm      meta-freescale-3rdparty  meta-nxp-demo-experience  meta-swupdate              meta-variscite-hab         poky
meta-browser  meta-freescale-distro    meta-openembedded         meta-timesys               meta-variscite-sdk-common  var-host-docker-containers
meta-clang    meta-imx                 meta-qt6                  meta-variscite-bsp-common  meta-variscite-sdk-imx
```