varigit/variscite-bsp-platform #25: Configuration default.xml of kirstone Branch does not build

telkamp, 2024-09-02:
In the kirkstone branch the repo manifest [default.xml](https://github.com/varigit/variscite-bsp-platform/blob/kirkstone/default.xml) points to [kirkstone-5.15.xml](https://github.com/varigit/variscite-bsp-platform/blob/kirkstone/kirkstone-5.15.xml). The last commit was from Jul 12, 2014. It checks out linux-variscite-4.14.78... and it does not build:
[image]

Instead, the repo manifest [kirkstone-5.15.71-2.2.0.xml](https://github.com/varigit/variscite-bsp-platform/blob/kirkstone/kirkstone-5.15.71-2.2.0.xml) works. Is [default.xml](https://github.com/varigit/variscite-bsp-platform/blob/kirkstone/default.xml) broken?

WERTBON1789, 2024-09-02:
Additional Info:
Branch: kirkstone (commit id: b8a6f0172f3969eca4ec35b48a53c3bbda465b6d)
used XML is default.xml which links to kirkstone-5.15.xml

Part of the output from `bitbake-layers show-recipes`:

```
...
linux-fslc:
  meta-freescale       5.18 (skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-variscite, not linux-fslc)
linux-fslc-imx:
  meta-freescale       5.15 (skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-variscite, not linux-fslc-imx)
linux-fslc-lts:
  meta-freescale       5.15 (skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-variscite, not linux-fslc-lts)
linux-fslc-lts-mfgtool:
  meta-freescale       5.15.48+gitAUTOINC+d14543c880
linux-fslc-mfgtool:
  meta-freescale       5.18.5+gitAUTOINC+1d6b3055ae
linux-fslc-qoriq:
  meta-freescale       5.4 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
linux-imx:
  meta-freescale       5.15 (skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-variscite, not linux-imx)
linux-imx-headers:
  meta-freescale       5.15
linux-imx-mfgtool:
  meta-freescale       5.15.52+gitAUTOINC+36363d8623
linux-kontron:
  meta-freescale-3rdparty 5.10 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
linux-libc-headers:
  meta                 5.16
linux-qoriq:
  meta-freescale       5.10 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
  meta-freescale-3rdparty 5.4 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
linux-toradex:
  meta-freescale-3rdparty 4.4 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
  meta-freescale-3rdparty 5.4-2.3.x (skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-variscite, not linux-toradex)
linux-variscite:
  meta-freescale-3rdparty 4.14.78+gitAUTOINC+0b8118cd4d
  meta-variscite-bsp   5.15 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
linux-yocto:
  meta                 5.10 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
  meta                 5.15 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
  meta                 5.15 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
linux-yocto-dev:
  meta                 unknown (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
linux-yocto-rt:
  meta                 5.10 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
  meta                 5.15 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
linux-yocto-tiny:
  meta                 5.10 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
  meta                 5.15 (skipped: incompatible with machine imx6ul-var-dart (not in COMPATIBLE_MACHINE))
...
```

Full Output: [bitbake-layers.txt](https://github.com/user-attachments/files/16840265/bitbake-layers.txt)

nsdrude, 2024-09-03:
@WERTBON1789 the DART-6UL is only supported by [kirkstone-5.15.71-2.2.0.xml](https://github.com/varigit/variscite-bsp-platform/blob/mx6ul-yocto-kirkstone-5.15.71_2.2.0-v1.1/kirkstone-5.15.71-2.2.0.xml)

Our wiki lists which releases are supported and has guides for building Yocto for reach: https://variwiki.com/index.php?title=DART-6UL

In general, you should do the following to make sure you're following the correct guide for your release:

1. Visit [variwiki.com](https://variwiki.com/index.php?title=Main_Page)
2. Select your SoM
3. Select the developers guide for your software release

In the future, please open a ticket in our customer support portal: https://www.variscite.com/support/

closed 2024-09-03 by nsdrude