varigit/variscite-bsp-platform #26: Debian build failed with Yocto

anderson1008, 2024-09-17:
I am following this recipe and build debian in docker. 
https://variwiki.com/index.php?title=Yocto_Debian_Build_Release&release=mx8mn-debian-bookworm-6.1.36_2.1.0-v1.0

Specifically, I am building based on mickledore branch.
```
$ repo init -u https://github.com/varigit/variscite-bsp-platform.git -b mickledore -m imx-6.1.36-2.1.0-debian.xml
$ repo sync -j$(nproc)
```

However, I am see this error during bitbake:
```
ERROR: Task (/workdir/debian_build/debian_09141100/sources/meta-variscite-bsp-imx/recipes-bsp/freertos-variscite/freertos-variscite_2.15.x.bb:do_compile) failed with exit code '1'
WARNING: Variable key RDEPENDS:${PN} ( attr) replaces original key RDEPENDS:coreutils ( ${@bb.utils.contains('PACKAGECONFIG', 'single-binary', '', 'coreutils-stdbuf', d)}).

```

nsdrude, 2024-09-17:
@anderson1008 please write the full steps to reproduce this, including how you start the docker container, what your host computer is, etc.

anderson1008, 2024-09-17:
[docker.zip](https://github.com/user-attachments/files/17034738/docker.zip)

Please see the attached scripts. Docker is started by doing
```
run.sh -p -u 22.04
```
My PC runs Ubuntu. 

Here is CPU spec
```
vendor_id	: GenuineIntel
cpu family	: 6
model		: 183
model name	: 13th Gen Intel(R) Core(TM) i7-13700F
stepping	: 1
microcode	: 0x123
cpu MHz		: 800.000
cache size	: 30720 KB
physical id	: 0
siblings	: 24
core id		: 39
cpu cores	: 16
apicid		: 78
initial apicid	: 78
fpu		: yes
fpu_exception	: yes
cpuid level	: 32
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdseed adx smap clflushopt clwb intel_pt sha_ni xsaveopt xsavec xgetbv1 xsaves split_lock_detect user_shstk avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req hfi vnmi umip pku ospke waitpkg gfni vaes vpclmulqdq rdpid movdiri movdir64b fsrm md_clear serialize arch_lbr ibt flush_l1d arch_capabilities
vmx flags	: vnmi preemption_timer posted_intr invvpid ept_x_only ept_ad ept_1gb flexpriority apicv tsc_offset vtpr mtf vapic ept vpid unrestricted_guest vapic_reg vid ple shadow_vmcs ept_mode_based_exec tsc_scaling usr_wait_pause
bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs eibrs_pbrsb rfds bhi
```

anderson1008, 2024-09-17:
I also run directly w/o docker. It produce the exact the same error.
[console-latest.log](https://github.com/user-attachments/files/17035502/console-latest.log)

nsdrude, 2024-12-18:
@anderson1008 I just ran a build from the beginning and it finished successfully.

closed 2024-12-18 by nsdrude