varigit/variscite-bsp-platform #27: Kirkstone Branch Build Fails Due to fmt library

Hassin10760, 2024-12-13:
I am Building the Kirkstone branch for var-som-mx8mp SOM version 2.0 
the Build fails with the following error 

| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h:1581:7: note: 'formattable' evaluates to false
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h: In instantiation of 'constexpr fmt::v10::detail::value<Context> fmt::v10::detail::make_arg(T&) [with bool PACKED = true; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; T = Fsl::VirtualMouseButton::Enum; typename std::enable_if<PACKED, int>::type <anonymous> = 0]':
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h:1810:51:   required from 'constexpr fmt::v10::format_arg_store<Context, Args>::format_arg_store(T& ...) [with T = {Fsl::VirtualMouseButton::Enum, bool, Fsl::PxPoint2}; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; Args = {Fsl::VirtualMouseButton::Enum, bool, Fsl::PxPoint2}]'
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h:1828:18:   required from 'constexpr fmt::v10::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<T>::type>::type ...> fmt::v10::make_format_args(T& ...) [with Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; T = {Fsl::VirtualMouseButton::Enum, bool, Fsl::PxPoint2}]'
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h:2790:44:   required from 'std::string fmt::v10::format(fmt::v10::format_string<T ...>, T&& ...) [with T = {Fsl::VirtualMouseButton::Enum, bool, Fsl::PxPoint2}; std::string = std::__cxx11::basic_string<char>; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, Fsl::VirtualMouseButton::Enum, bool, Fsl::PxPoint2>]'
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git/DemoApps/Shared/System/InputEvents/source/Shared/System/InputEvents/InputLogger.cpp:71:27:   required from here
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h:1577:63: error: 'fmt::v10::detail::type_is_unformattable_for<Fsl::VirtualMouseButton::Enum, char> _' has incomplete type
|  1577 |     type_is_unformattable_for<T, typename Context::char_type> _;
|       |                                                               ^
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h:1581:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
|  1581 |       formattable,
|       |       ^~~~~~~~~~~
| /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/recipe-sysroot/usr/include/fmt/core.h:1581:7: note: 'formattable' evaluates to false
| [101/2889] Building CXX object DemoApps/Shared/System/DFGraphicsBasic2D/CMakeFiles/Shared.System.DFGraphicsBasic2D.dir/workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git/DemoApps/Shared/System/DFGraphicsBasic2D/source/Shared/System/DFGraphicsBasic2D/Shared.cpp.o
| [102/2889] Building CXX object DemoApps/Shared/System/OnDemandRendering/CMakeFiles/Shared.System.OnDemandRendering.dir/workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git/DemoApps/Shared/System/OnDemandRendering/source/Shared/System/OnDemandRendering/Chart/CustomChartGridLines.cpp.o
| [103/2889] Building CXX object DemoApps/Shared/System/OnDemandRendering/CMakeFiles/Shared.System.OnDemandRendering.dir/workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git/DemoApps/Shared/System/OnDemandRendering/source/Shared/System/OnDemandRendering/JankDetector.cpp.o
| [104/2889] Building CXX object DemoApps/Shared/System/DFNativeBatch2D/CMakeFiles/Shared.System.DFNativeBatch2D.dir/workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git/DemoApps/Shared/System/DFNativeBatch2D/source/Shared/System/DFNativeBatch2D/TextureAtlasScene.cpp.o
| [105/2889] Building CXX object DemoApps/Shared/System/DevNativeTexture2D/CMakeFiles/Shared.System.DevNativeTexture2D.dir/workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git/DemoApps/Shared/System/DevNativeTexture2D/source/Shared/System/DevNativeTexture2D/Shared.cpp.o
| ninja: build stopped: subcommand failed.
| WARNING: The build command 'cmake --build . --target install -- -j 6' failed with '1'. It was run with CWD: '/workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git/build/Yocto/Ninja/release'
| WARNING: /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/temp/run.do_compile.1726185:144 exit 1 from 'FslBuild.py -vvvvv -t sdk --UseFeatures [EarlyAccess,EGL,GoogleUnitTest,Lib_NlohmannJson,OpenVG,G2D,OpenGLES2,OpenCV4,Vulkan1.2,OpenGLES3.2,OpenCL1.2,OpenVX1.1] --UseExtensions [OpenGLES3:GL_EXT_color_buffer_float] --Variants [WindowSystem=Wayland] --BuildThreads 6 -c install --CMakeInstallPrefix /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/git'
| WARNING: Backtrace (BB generated script):
| 	#1: do_compile, /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/temp/run.do_compile.1726185, line 144
| 	#2: main, /workdir/build_xwayland/tmp/work/armv8a-mx8mp-fslc-linux/imx-gpu-sdk/6.0.2-r0/temp/run.do_compile.1726185, line 148
ERROR: Task (/workdir/sources/meta-freescale-distro/recipes-graphics/imx-gpu-sdk/imx-gpu-sdk_6.0.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2618 tasks of which 2087 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 8 seconds

Attached is the local.conf 
[image]

bonaldig, 2024-12-16:
Hi @Hassin10760 !

I've tried to reproduce but I didn't have success. These were the steps I've followed:

1.  Prepare yocto environment with the following commands:
```
cd ~/
git clone https://github.com/varigit/var-host-docker-containers.git
mkdir -p yocto/8mp-kirkstone && cd yocto/8mp-kirkstone
repo init -u https://github.com/varigit/variscite-bsp-platform.git -b kirkstone -m kirkstone-5.15.71-2.2.0.xml
repo sync -j$(nproc)
../var-host-docker-containers/run.sh -u 20.04 -w ~/yocto/
cd 8mp-kirkstone
MACHINE=imx8mp-var-dart DISTRO=fslc-xwayland . var-setup-release.sh build_xwayland
```
2. Modify conf/local.conf to set the following variables:
-    DL_DIR ?= "/workdir/downloads/"
-    SSTATE_DIR ?= "/workdir/sstate/" 
3. Build fsl-image-gui:
`bitbake fsl-image-gui`

Could you try those steps?

Hassin10760, 2024-12-19:
Hi @bonaldig After running the build with Ubuntu 20.04 container I am facing the following problem. I would be glad if you kindly have a look into it 

`Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mp-var-dart"
DISTRO               = "fslc-xwayland"
DISTRO_VERSION       = "4.0"
TUNE_FEATURES        = "aarch64 armv8a crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:6505459809380ddcf152a09343e4dc55038de332"
meta-oe              
meta-multimedia      
meta-python          
meta-filesystems     
meta-gnome           
meta-networking      = "HEAD:82c75b466e55d7dca7a2364986ecb704cf63d141"
meta-freescale       = "HEAD:d9f77788e1902a7d999958575a955e701c1c1b61"
meta-freescale-3rdparty = "HEAD:f6fa0fd8783ce69d07feaad0b7ca6759b5a4d5d6"
meta-freescale-distro = "HEAD:d5bbb487b2816dfc74984a78b67f7361ce404253"
meta-freescale-ml    = "HEAD:e86f4ac6ed3c18c1ed3dd76824fda4dd0e8c4f8d"
meta-clang           = "HEAD:c728c3f9168c8a4ed05163a51dd48ca1ad8ac21d"
meta-qt5             = "HEAD:aec1934cb2708236c033f2d205d5c3fbbebf7555"
meta-python2         = "HEAD:f02882e2aa9279ca7becca8d0cedbffe88b5a253"
meta-swupdate        = "HEAD:91de63c006e886eba8d9c263c9aa55d4c7632072"
meta-virtualization  = "HEAD:a7413c5d7568ce91b809ed11f84305b1afb468bb"
meta-variscite-bsp   = "HEAD:5e42344c2bf6616c7270ff577e96b5bf5deacb0f"
meta-variscite-sdk   = "HEAD:63e583098e1af460d24bc2f816751c8c88f0218d"
meta-variscite-hab   = "HEAD:aee9a7a30d133861b21bdab441d0c91828f6f690"

NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 4 seconds

Summary: There was 1 ERROR message, returning a non-zero exit code.
Loading cache: 100% |###############################################################################################################################################################################| Time: 0:00:01
Loaded 5213 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mp-var-dart"
DISTRO               = "fslc-xwayland"
DISTRO_VERSION       = "4.0"
TUNE_FEATURES        = "aarch64 armv8a crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:6505459809380ddcf152a09343e4dc55038de332"
meta-oe              
meta-multimedia      
meta-python          
meta-filesystems     
meta-gnome           
meta-networking      = "HEAD:82c75b466e55d7dca7a2364986ecb704cf63d141"
meta-freescale       = "HEAD:d9f77788e1902a7d999958575a955e701c1c1b61"
meta-freescale-3rdparty = "HEAD:f6fa0fd8783ce69d07feaad0b7ca6759b5a4d5d6"
meta-freescale-distro = "HEAD:d5bbb487b2816dfc74984a78b67f7361ce404253"
meta-freescale-ml    = "HEAD:e86f4ac6ed3c18c1ed3dd76824fda4dd0e8c4f8d"
meta-clang           = "HEAD:c728c3f9168c8a4ed05163a51dd48ca1ad8ac21d"
meta-qt5             = "HEAD:aec1934cb2708236c033f2d205d5c3fbbebf7555"
meta-python2         = "HEAD:f02882e2aa9279ca7becca8d0cedbffe88b5a253"
meta-swupdate        = "HEAD:91de63c006e886eba8d9c263c9aa55d4c7632072"
meta-virtualization  = "HEAD:a7413c5d7568ce91b809ed11f84305b1afb468bb"
meta-variscite-bsp   = "HEAD:5e42344c2bf6616c7270ff577e96b5bf5deacb0f"
meta-variscite-sdk   = "HEAD:63e583098e1af460d24bc2f816751c8c88f0218d"
[... 4 lines cut ...]
NOTE: Executing Tasks
WARNING: assimp-5.0.1-r0 do_fetch: Failed to fetch URL git://github.com/assimp/assimp.git;protocol=https;branch=assimp_5.0_release, attempting MIRRORS if available
WARNING: devil-1.8.0-r0 do_fetch: Failed to fetch URL http://sourceforge.net/projects/openil/files/DevIL/1.8.0/DevIL-1.8.0.zip, attempting MIRRORS if available
ERROR: devil-1.8.0-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/workdir/build_xwayland/tmp/sysroots-uninative/x86_64-linux/usr/bin:/workdir/sources/poky/scripts:/workdir/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/devil/1.8.0-r0/recipe-sysroot-native/usr/bin/aarch64-fslc-linux:/workdir/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/devil/1.8.0-r0/recipe-sysroot/usr/bin/crossscripts:/workdir/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/devil/1.8.0-r0/recipe-sysroot-native/usr/sbin:/workdir/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/devil/1.8.0-r0/recipe-sysroot-native/usr/bin:/workdir/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/devil/1.8.0-r0/recipe-sysroot-native/sbin:/workdir/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/devil/1.8.0-r0/recipe-sysroot-native/bin:/workdir/sources/poky/bitbake/bin:/workdir/build_xwayland/tmp/hosttools"; export HOME="/home/vari"; /usr/bin/env wget -t 2 -T 30 --passive-ftp -P /workdir/downloads/ 'http://sourceforge.net/projects/openil/files/DevIL/1.8.0/DevIL-1.8.0.zip' --progress=dot -v failed with exit code 4, no output
ERROR: devil-1.8.0-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'http://sourceforge.net/projects/openil/files/DevIL/1.8.0/DevIL-1.8.0.zip')
ERROR: Logfile of failure stored in: /workdir/build_xwayland/tmp/work/imx8mp_var_dart-fslc-linux/devil/1.8.0-r0/temp/log.do_fetch.2453019
ERROR: Task (/workdir/sources/meta-freescale-distro/recipes-graphics/devil/devil_1.8.0.bb:do_fetch) failed with exit code '1'
[... cut ...]
NOTE: Tasks Summary: Attempted 9943 tasks of which 665 didn't need to be rerun and 2 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 16 seconds

Summary: 2 tasks failed:
  /workdir/sources/meta-freescale-distro/recipes-graphics/devil/devil_1.8.0.bb:do_fetch
  /workdir/sources/meta-freescale-ml/recipes-libraries/deepview-rt/deepview-rt_2.4.46-aarch64.bb:do_package_qa
Summary: There were 6 WARNING messages.
Summary: There were 4 ERROR messages, returning a non-zero exit code.
`

Hassin10760, 2024-12-27:
Dear @bonaldig on a fresh environment I executed your steps and the build was successful . thanks for the support . 
closing this issue

closed 2024-12-27 by Hassin10760