varigit/variscite-bsp-platform #31: Error with Yocto 5.0 Scarthgap

antonio-fr, 2025-03-25:
Using this repo with
-b scarthgap -m imx-6.6.52-2.2.0.xml

That is :
https://github.com/varigit/variscite-bsp-platform/blob/scarthgap/imx-6.6.52-2.2.0.xml

I got the following error when bitbaking the fsl-image-gui.
```
freertos-variscite-2.15.x-r0 do_fetch: Fetcher failure: Unable to find revision f617456ed00fc750ab1dea844d2f8ea66f9fa7aa in branch mcuxpresso_sdk_2.15.x-var01 even from upstream
freertos-variscite-2.15.x-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/varigit/freertos-variscite.git;protocol=https;branch=mcuxpresso_sdk_2.15.x-var01;')
```

That breaks the building of Yocto 5.0 on iMX8MP.

Looks like the issue is because f617456ed00fc750ab1dea844d2f8ea66f9fa7aa commit is part of mcuxpresso_sdk_2.15.x-var02 (not ...var01).
I can see [here](https://github.com/varigit/meta-variscite-bsp-imx/commit/ccbb4a0e722d4aa3600f17d684a990bb9f7c8617) this is aligned, but Idk, at some point it is confused.

I have multiple times, trying to reset all I can, removing cahche, etc.. but no luck.

antonio-fr, 2025-03-25:
After starting from scratch on a new machine, I confirm the issue. There's something off with the commits you did today.

```
$ repo init -u https://github.com/varigit/variscite-bsp-platform.git -b scarthgap -m imx-6.6.52-2.2.0.xml
$ repo sync
$ MACHINE=imx8mp-var-dart DISTRO=fsl-imx-xwayland . var-setup-release.sh build_xwayland
$ bitbake fsl-image-gui --runall=fetch
...
ERROR: freertos-variscite-2.15.x-r0 do_fetch: Fetcher failure: Unable to find revision f617456ed00fc750ab1dea844d2f8ea66f9fa7aa in branch mcuxpresso_sdk_2.15.x-var01 even from upstream
ERROR: freertos-variscite-2.15.x-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/varigit/freertos-variscite.git;protocol=https;branch=mcuxpresso_sdk_2.15.x-var01;')
ERROR: Task failed
```

For the moment, I'll use the version from some days ago. https://github.com/varigit/variscite-bsp-platform/commit/c3e9dfbf6ad707bd85445fb6a11e531a2de66675

antonio-fr, 2025-03-25:
FYI, using the previous version from March 21st, it works. This confirm the recent changes messed up somewhere in the dependency tree.
```
$ repo init -u https://github.com/varigit/variscite-bsp-platform.git -b c3e9dfbf6ad707bd85445fb6a11e531a2de66675 -m imx-6.6.52-2.2.0.xml
...
$ bitbake fsl-image-gui --runall=fetch
...
Tasks all succeeded.
```

nsdrude-var, 2025-03-25:
@antonio-fr thanks for reporting this. We'll fix this within 24 hours.

varigigi, 2025-03-26:
@antonio-fr, please refresh your manifest running `repo sync` and start a new build.
The problem has been fixed [here](https://github.com/varigit/meta-variscite-bsp-imx/commit/33522ae7ce0d).

antonio-fr, 2025-03-26:
I confirm this is fixed. Thanks for the quick fix! 🚀 👍

closed 2025-03-26 by antonio-fr