varigit/variscite-bsp-platform #32: backport d5bf5e601bd24fb03bfc2342f1b0a41b3f0bddb4  to dunfell branch

m-tmatma, 2025-04-12:
Please backport d5bf5e601bd24fb03bfc2342f1b0a41b3f0bddb4  to dunfell branch.

This commit is useful for other branchs.

morishitaandre, 2025-04-14:
Hi @m-tmatma,

Thanks for reporting this. This will be fixed shortly.

Best regards,
André

morishitaandre, 2025-04-15:
Hi @m-tmatma,

dunfell was just updated: https://github.com/varigit/variscite-bsp-platform/commits/dunfell/

A `repo sync` should update your manifest. Can you please check?

m-tmatma, 2025-04-16:
@morishitaandre 
Thank you.
It worked.

These repos rely on https://git.yoctoproject.org/git too.

https://github.com/varigit/variscite-bsp-platform/blob/fe3dd1f094a2ec4dac76e30ccdb825149cd4dfab/default.xml#L15

https://github.com/varigit/variscite-bsp-platform/blob/fe3dd1f094a2ec4dac76e30ccdb825149cd4dfab/default.xml#L32

But there are no mirrors for these repositories at https://github.com/yoctoproject

How about forking them on https://github.com/varigit and using the mirrors too.

nsdrude-var, 2025-04-16:
@m-tmatma thanks for the suggestion, we've considered it but for now do not plan to create mirrors for those repositories.

closed 2025-04-16 by nsdrude-var

m-tmatma, 2025-04-16:
@nsdrude-var 
Thank you for replying.

If possible, tell me why you have dicided not to create mirrors for my reference.

m-tmatma, 2025-04-19:
@nsdrude-var 
Cloning those repository may be fast from US.
But it is very slow from Japan.

This is GitHub Actions job to clone those repositories.
https://github.com/m-tmatma/test-clone-speed/actions/runs/14549618576/job/40819232149

cloning https://git.yoctoproject.org/git/meta-freescale takes only 2 seconds.
cloning https://git.yoctoproject.org/git/meta-virtualization takes only 1 seconds.

GitHub Actions may be run in US.

But cloning https://git.yoctoproject.org/git/meta-freescale takes 2 minutes.

```
 $ time git clone https://git.yoctoproject.org/git/meta-freescale
Cloning into 'meta-freescale'...
remote: Enumerating objects: 54039, done.
remote: Counting objects: 100% (5848/5848), done.
remote: Compressing objects: 100% (792/792), done.
remote: Total 54039 (delta 5504), reused 5053 (delta 5053), pack-reused 48191
Receiving objects: 100% (54039/54039), 48.31 MiB | 414.00 KiB/s, done.
Resolving deltas: 100% (32492/32492), done.

real    2m0.586s
user    0m2.573s
sys     0m1.120s
```

cloning https://git.yoctoproject.org/git/meta-virtualization is OK.

```
 $ time git clone https://git.yoctoproject.org/git/meta-virtualization
Cloning into 'meta-virtualization'...
remote: Enumerating objects: 19488, done.
remote: Counting objects: 100% (228/228), done.
remote: Compressing objects: 100% (202/202), done.
remote: Total 19488 (delta 85), reused 47 (delta 24), pack-reused 19260
Receiving objects: 100% (19488/19488), 6.47 MiB | 396.00 KiB/s, done.
Resolving deltas: 100% (12149/12149), done.

real    0m17.886s
user    0m0.769s
sys     0m0.209s
``

The purpose to create mirrors is to speed up cloning repository.

nsdrude-var, 2025-04-21:
@m-tmatma any chance it's your local network? From both the US and Israel the speeds are similiar:

```
$ time git clone https://git.yoctoproject.org/git/meta-virtualization
Cloning into 'meta-virtualization'...
remote: Enumerating objects: 19513, done.
remote: Counting objects: 100% (98/98), done.
remote: Compressing objects: 100% (98/98), done.
remote: Total 19513 (delta 27), reused 0 (delta 0), pack-reused 19415
Receiving objects: 100% (19513/19513), 6.41 MiB | 8.69 MiB/s, done.
Resolving deltas: 100% (12205/12205), done.

real	0m1.652s
user	0m0.509s
sys	0m0.151s

$ time git clone https://git.yoctoproject.org/git/meta-freescale
Cloning into 'meta-freescale'...
remote: Enumerating objects: 54039, done.
remote: Counting objects: 100% (5923/5923), done.
remote: Compressing objects: 100% (798/798), done.
remote: Total 54039 (delta 5580), reused 5122 (delta 5122), pack-reused 48116
Receiving objects: 100% (54039/54039), 48.31 MiB | 19.84 MiB/s, done.
Resolving deltas: 100% (32490/32490), done.

real	0m3.088s
user	0m1.592s
sys	0m0.505s
```